betaflight/betaflight-configurator #292: 3D model animation defaulting to generic box - OS X

etracer65, 2016-09-21:
Not sure when this happened (but sometime in the past week), but on my MacBook Pro (OS X 10.10.5) the model animation is showing the generic Cleanflight box instead of the quad.  It's like what happens if you run a custom mixer (but I'm not).  Tried a complete refresh (w/chip erase) to BF 3.0.0, uninstalled and reinstalled Chrome, removed and reinstalled the BF configurator (1.8.1) - all with no change.

The only clue I can see is in the Chrome console I see the message that SVG animations are deprecated.

[image]
[image]

mikeller, 2016-09-21:
This is weird.The warning message is unrelated, and has always been there. Only thing I can imagine is that it might have been caused by an update to chrome, or that the uninstall of chrome and / or the configurator did leave behind some artefacts that broke the reinstall (a common problem on Windows, and I am not a Mac user, so I don't know if it can happen there too).

chipkaye, 2016-09-26:
Line 49 in README.md mentions this for Linux users, but I am running OS X 10.11.6, had the same problem and enabling this switch in Chrome fixed the problem:

"If you have 3D model animation problems, enable "Override software rendering list" in Chrome flags chrome://flags/#ignore-gpu-blacklist"

mikeller, 2016-09-26:
@chipkaye: Thanks, amended `README.md`.

etracer65, 2016-09-26:
I can confirm this fixed it for me as well.  Tried everything prior to this including upgrading to the latest OS X 10.12 and trying the DEV channel for Chrome.  It looks like there were Chrome desktop releases on 13-Sep and 14-Sep and this fits into the timeframe when it stopped working for me.

mikeller, 2016-09-26:
Yeah, I guess this is one of the downsides of using Chrome as a distribution channel...

macoces, 2017-06-06:
Fixed! Thanks to "chipkaye" :)

AndersHoglund, 2017-06-06:
Not only Linux and OSX, I had the very same problem on Win10 and I can confirm this fix worked just fine here too. General OS independent Chrome problem it seems.

socialmedialabs, 2018-01-31:
While the fix @chipkaye proposed works great with the Chrome extension it doesn't work in the current standalone version. Any way to apply this fix to the latest version as well?

etracer65, 2018-01-31:
Yeah, I saw this as well on my (relatively) old MacBook Pro - reverting to the generic box instead of the 3D model.

closed 2018-05-13 by stale[bot]