betaflight/betaflight-configurator #299: Mixer QuadX1234 not selectable

dropax, 2016-09-24:
The option "QuadX1234" is not in the dropdown list but still can be set in CLI with the mixer command. 
I recall it was in the GUI back then we used cleanflight configurator. Very handy when converting a CC3D quad from openpilot to betaflight. 

Is it betaflight configurator not showing or betaflight not telling via MSP? Had a quick glance at both but couldn't spot anything.

mikeller, 2016-09-25:
It looks like it is a configurator problem. `QuadX1234` should be the last entry in `mixerList` in `js/model.js`, right below `Custom Tricopter`. But it could also be an oversight in the CLI code that this mixer type has never been removed from the CLI when support for it was removed from the firmware.

Can you please confirm if this still works as it should when configured from the CLI?

dropax, 2016-09-26:
It is still in https://github.com/betaflight/betaflight/blob/master/src/main/flight/mixer.c
Can't confirm it working personally (my only CC3D in a quad doesn't boot atm, guess it's offended of being ignored for too long ;)) but others had success with my advice.

mikeller, 2016-09-26:
Fair enough, I can readd it to the list. Just beat with me, I'm on holiday at the moment.

PR #314 (2016-10-12): Added 'Quad X 1234' model back into model list.
Fixes #299.
closed 2016-10-12 by borisbstyle