betaflight/betaflight-configurator #303: Load backup without connecting to board for review

Hawkers, 2016-09-29:
I think it would be helpful to open the interface up when not connected to a board.  For example when somebody shares their backup with me, I'd like to look it over without plugging my board in.

mikeller, 2016-09-29:
I've thought about this before, and it would be a very useful feature indeed. 3.1 will introduce a completely new way of how the configuration is represented, and the right thing to do is to wait with this feature and then implement ort based on the new configuration format. Doing it now would mean basing it on something that is due to be deprecated in 3.1.

Hawkers, 2016-09-29:
This is good to know.  I'm the developer at RotorBuilds.com and plan to implement BF backup uploads to let users share their rates and PIDs.  If the backup file format is changing I may want to wait until 3.1 is released before I write my parser.  Will it still be in the json format?  Will a CLI "dump" still look essentially the same?

mikeller, 2016-10-09:
It will still be json, but the structure of the json will most likely change, since the json is nothing but a dump of the internal data structures that are used to hold the FC configuration in the configurator.

CLI `dump` line ordering will likely change too, but probably in an easier to deal with fashion, since the individual settings will still be called the same, so line by line parsing will still work.

haslinghuis, 2021-06-05:
@Hawkers should be fixed in https://github.com/betaflight/betaflight-configurator/pull/2283

closed 2021-06-05 by haslinghuis