betaflight/betaflight-configurator #309: Problem with OSD configuration

sebastianha, 2016-10-05:
I enabled OSD in the configurator but I cannot use the OSD tab as it is not properly displayed (see screenshot):

[image]

I am running betaflight 3.0.0 with configurator 1.8.1 (master had the same problem) on a naze32 and a micro-minimosd with the latest software. The OSD is connected through soft-serial and running a 19200 baud.

Can anyone reproduce this problem?

nathantsoi, 2016-10-05:
Yeah, that's not going to work. The osd tab only works with the betaflight
osd (max7456 integrated via spi)

I'm surprised the osd tab was visible at all?

sebastianha, 2016-10-05:
Ah, ok. I thought it will also work with minimosd. I had to enable OSD to use the OSD-Mode (see #251)

Thanks for the quick answer!

closed 2016-10-05 by sebastianha

mikeller, 2016-10-05:
@nathantsoi : 

> I'm surprised the osd tab was visible at all?

I think this is caused by configurator not understanding the format of the OSD related MSP commands for firmware that was compiled without OSD support, and throwing exceptions. Probably better make the commands entirely conditional in the firmware (giving you 'unknown MSP command' instead of an exception), or make the configurator understand the syntax.

mactac, 2017-07-26:
I am having this exact problem, however I am using a betaflight F3 board.   My computer displays fine, but my android tablet does not, it displays exactly as the one above.