betaflight/betaflight-configurator #317: Not able to Config OSD

FlyinMikey, 2016-10-14:
I'm not able to configure my OSD with betaflight. Quadcopter is the Spedix S250 PRO and it has a Naze Rev6 with built in(I think) MinimOSD.  Is there some setting I need to enable?  All I see is Elements, Preview and Video format but there is no picture or display underneath.  Have tried with and without the battery connected.  Recently switched from cleanflight to betaflight 3.0 and Betaflight 1.8.3 configurator if it helps. Is there a comparability issue.  Any info would be great.  Thanks

mikeller, 2016-10-14:
Are you trying to use the 'OSD' tab in the Betaflight configurator to configure your minimOSD? This won't work, as the OSD tab does not (yet) support the minimOSD hardware.

FlyinMikey, 2016-10-14:
Thank you for the reply. When do you think the update will be available?

Mike N

mikeller, 2016-10-14:
@FlyinMikey: I don't think there is work being done (or even planned) to add support for minimOSD, the '(yet)' was more in the sense of 'I won't say it's not possible'. Sorry for that, but feel free to add it if you want to contribute.

closed 2016-10-25 by mikeller