betaflight/betaflight-configurator #327: Add 'Update Firmware' button to top bar, change 'Connect' button text to 'Configure'

mikeller, 2016-10-25:
This is to give users a more intuitive way to get into the firmware flasher, and remove the confusion around the 'Connect' button being visible when in the firmware flasher.

skaman82, 2016-10-25:
I think just renaming "Firmware flasher" into "Update Firmware" and hiding the "Connect" button would be better. I don't think that moving the buttons in the topbar is a good idea.

mikeller, 2016-10-25:
Why do you think it is not a good idea?

In general, having to change tabs on order to update firmware, and having to click 'Connect' in order to configure the FC seems to be inconsistent. Hiding the 'Connect' button while in the firmware flasher tab will just force users to first click on some other tab and then click 'Connect', in order to get to the configuration, so it actually makes the user experience worse.

skaman82, 2016-10-25:
@mikeller The connect button is pretty much the most important und most used button in the configurator. The firmware flasher not so much and having the two buttons side by side gives it the same importance in the GUI and that is why I think that is not such a good idea.

mikeller, 2016-10-25:
I can see your, but I think having the two main functions side by side doesn't detract from that (and the 'Configure' button stays in the same place for existing users).

Also, getting more users to regularly check for the availability of firmware updates would help reduce the number of 'why do i not get feature X?' => 'because you need to update your firmware.' support requests.

Would your concern be addressed by making the 'Update Firmware' button smaller (or less prominently coloured) than the 'Configure' button?

skaman82, 2016-10-26:
The connect button is pretty much the most important und most used button in the configurator. 
The firmware flasher not so much and having the two buttons side by side gives it the same importance in the GUI and that is why I think that is not such a good idea.

AndersHoglund, 2016-10-26:
Comeon. Try to look at the gui as a noob. The layout of buttons and tabs are not very user friendly.
I know a few that selected the "Firmware Flasher" and then hit the "Connect" button. And get stumped by what happens..... 
If one function is more important or not, or how frequently used, has nothing to do with the case. There has to be some clear and unambiguous selection between the two use cases, Config and Firmware update. Two equally important functions, if we should even discuss importance.

haslinghuis, 2024-05-06:
Fixed in #1203

closed 2024-05-06 by haslinghuis