betaflight/betaflight-configurator #330: 'Flash on connect' stops working after early disconnect

mikeller, 2016-10-26:
When flashing a bricked board by using 'No reboot sequence' / 'Flash on connect' and the boot pins, if a reboot sequence that fails to jump into the boot loader is interrupted by the board being disconnected before its end, subsequent reboot sequences will fail with 'previous device still flashing, please replug to try again'. This can only be reset by restarting the configurator.

borisbstyle, 2016-10-26:
not a high prio at all. It is advised everywhere to not disconnect your board during flash.

mikeller, 2016-10-26:
It's not during flash. It is when the flash never starts, because the boot loader pads deep inside the craft are not properly bridged. This should not cause the configurator to hang.

mikeller, 2016-10-26:
But if this is not a priority, we should remove this functionality, since it is clearly broken.

closed 2018-05-06 by mikeller