betaflight/betaflight-configurator #331: Strange defaults in Motor Test when DSHOT is used.

AndersHoglund, 2016-10-26:
Referring to betaflight PR 1389, there are still a few problems in BFC motor test. 
If DSHOT150 is enabled and min_command is set to 990 the defaults in the motor tab is a bit off.
1. Motor Test enabled by default.
2. Throttle min values follow min_command for inactive motors.
3. Throttle min values follow min_command if slider is moved for active motors. Default OK.

Nb. Only tested with DSHOT150, no tests with DSHOT300 nor DSHOT600. 

[image]

mikeller, 2016-10-26:
Will look into fixing these once the internal DShot data format has been finalised, and a decision made re adding an 'arm' command. (No point allowing 'disarmed' input from configurator if the ESCs have to be armed to get them to run.)

brucesdad13, 2017-04-04:
This is happening for me also with brushed I went to motors tab. If a battery is plugged in immediately motors all go full throttle and I can't turn them off even unplugging the USB the motors remain and I have to pull the LiPO. Without LiPO the flight controller has a brownout and reboots.

mikeller, 2017-04-05:
The original bug had been fixed in the meantime, but at the moment the motor tab is broken because it's missing some changes that have been made in cleanflight, but not yet ported to Betaflight.

brucesdad13, 2017-04-05:
Ok thanks for clarifying :)

closed 2018-05-13 by stale[bot]