betaflight/betaflight-configurator #333: 3D model animation defaulting to generic box - Windows 7

Copter38, 2016-10-30:
I recently (couple days or so) bounced into the exact same issue as described by etracer65.
The model animation is showing the generic Cleanflight box instead of the quad and this, of course, is with a standard Quad mix.

Have tried the fix proposed by chipkaye of changing the "Override software rendering list" flag in Chrome but to no luck. I can also confirm that reinstalling the 1.8.1 version of BF Configurator does not bring back the 3D quad animation...

What else could be causing this quite annoying issue?

mikeller, 2016-10-30:
Most likely it's something to do with a Chrome update. You could try to uninstall / reinstall Chrome.

Copter38, 2016-10-30:
I have fully uninstalled Chrome and reinstalled latest version of Chrome. Still no luck!

Copter38, 2016-12-20:
The issue disappeared as it came. Now the 3D model is showing properly again :)
I guess it really had something to do with a Chrome update.

closed 2016-12-20 by Copter38