betaflight/betaflight-configurator #341: Full chip erase on targets with large flash fail

blckmn, 2016-11-17:
@mikeller its actually the chip erase takes longer than the USB time out. The chip erase doesn't actually take too long (a few seconds) BUT in stm32usbdfu.js

```            if (self.options.erase_chip) {
                // full chip erase
                console.log('Executing global chip erase');
                $('span.progressLabel').text('Erasing ...');

                self.controlTransfer('out', self.request.DNLOAD, 0, 0, 0, [0x41], function () {
                    self.controlTransfer('in', self.request.GETSTATUS, 0, 0, 6, 0, function (data) {
                        if (data[4] == self.state.dfuDNBUSY) { // completely normal
                            var delay = data[1] | (data[2] << 8) | (data[3] << 16);
                            setTimeout(function () {
                                self.controlTransfer('in', self.request.GETSTATUS, 0, 0, 6, 0, function (data) {
                                    if (data[4] == self.state.dfuDNLOAD_IDLE) {
                                        self.upload_procedure(4);
                                    } else {
                                        console.log('Failed to execute global chip erase');
                                        self.upload_procedure(99);
                                    }
                                });
                            }, delay);
                        } else {
                            console.log('Failed to initiate global chip erase');
                            self.upload_procedure(99);
                        }
                    });
                });```

In the case 2, the self.request.GETSTATUS comes back with the data undefined. This is because the USB times out - whilst the chip is erasing there are no packets. Increasing the value of "delay" doesn't work. However if you set a break point and pause for 3-4 seconds, it will actually work when you start running again.

I thought about changing setTimeout to setInterval and continuously checking until the DFU was idle (or a counter exceeded acceptable limits) but my javascript is not good enough.

marcroe, 2016-11-17:
I could have a look at this. There seems to be a bug/problem with the crome USB API preventing the USB timeout from working properly on some platforms/OSes. Full chip erase can take up to 18 seconds on some boards, as measured with my modified configurator under https://github.com/marcroe/betaflight-configurator.

blckmn, 2016-11-17:
Thanks. If we could allow for 20 seconds and update the progress bar say 5% each second, going to 100% when idle... or overall timeout.

Just to show some progress to user.

mikeller, 2016-11-18:
Yeah, I've seen some cases in JS where OS calls caused timeouts to run short. One possible workaround is to note time before the call, and when the timeout callback is called compare time to ascertain that the timeout has really elapsed, otherwise reschedule it.

Happy for you to have a look at this, please submit a PR if you have got a solution.

I am not seeing this problem with my BLUEJAYF4 on ubuntu, and I am not terribly keen on trying to debug this with USB passthrough to Windows running inside VirtualBox.

mikeller, 2016-11-19:
Fixed by #344.

closed 2016-11-19 by mikeller