betaflight/betaflight-configurator #354: Feature request: Touch friendly interface

danarrib, 2016-11-30:
Hello. I use Betaflight configurator in the field, and I have a Windows 10 tablet (Full Windows, X86 processor), but I do not have mouse or keyboard. I use only touchscreen and on-screen keyboard that takes half of the screen.

I was wondering if it's possible to rethink the UI to a "responsive" one, that is touch friendly, with larger interface components, and without textboxes... Because every time a textbox is focused, the onscreen keyboard appears and brakes everything.

That's it. I can also help as I am a web developer.
What do you think?
Thanks.

mikeller, 2016-11-30:
Hi @danarrib. There is an effort under way to remake the configurator into a 'next generation' version that will be having a more multi - platform approach, and won't be limited to a chrome app any more. Part of this will be looking into mobile support.

In the meantime, if you want to contribute and have suggestions on how to make the configurator more mobile friendly, please feel free to bring your suggestions up for discussion here, or even make them into a pull request. We are always grateful for community contributions.

nathantsoi, 2016-12-01:
Who is making this 'next generation' configurator @mikeller? Or where is development happening?

mikeller, 2016-12-02:
It's a community effort, like all of Betaflight, @nathantsoi. And it's in https://github.com/betaflight/betaflight-configurator-ng .

nathantsoi, 2016-12-02:
sweet!

closed 2018-05-13 by stale[bot]