betaflight/betaflight-configurator #356: Feature request:the design of multi-language support, English, Japanese, Korean,Chinese

Linjieqiang, 2016-12-01:
Hello guys,I have a idea.Why not support multi-language in betaflight-configurator.I can provide the Chinese translation of documents and pull request on this repertory.As I know,the chrome can support multi-language perfectly.So how do you think this idea?Anyway,thanks your great work for bf development.

Wind0r, 2016-12-01:
The configurator already "supports" some kind of multi-language, but currently there is only english.
https://github.com/betaflight/betaflight-configurator/tree/master/_locales

Also there isnt any way to change the language, but it should be hard to add,
but the big Problem is that multi-language support isn't just simply replacing some string.
Also on fast moving Projects some translation may get not updated and can create strange interfaces for the enduser and multi-language can create a large overhead on such a project.

But since this project is open source you can create a translation file and create a pull request :)

mikeller, 2016-12-01:
I think the way this works for Chrome apps is that chrome uses the language that matches the one that is set in Chrome automatically, if available.

And yes, while we are not specifically putting effort into getting the configurator translated, it can be done easily enough by translating `messages.json`, and we'd happily accept community contributions for this.

One thing that I think we'd probably want to do if we start offering translated versions is to add a disclaimer starting that support requests should be done in English.

Linjieqiang, 2016-12-01:
OK,I know your meaning.

closed 2016-12-13 by Linjieqiang