betaflight/betaflight-configurator #385: Yaw RC Rate forced to zero, MaxVel = NaN

riquezjp, 2017-01-11:
Using bf2.9.0 & Config: 1.8.7 
Yaw PID RC Rate is broken.
It is set to 0.00 & cannot be changed. PID graph doesnt show due to this error.
Max Vel cannot be calculated.
[image]

borisbstyle, 2017-01-11:
Please upgrade to 3.0 or 3.1 version. 

Older versions than 3.0 will not be maintained in the configurator. Especially the yaw rc rate was a workaround to get it working.

closed 2017-01-11 by borisbstyle

riquezjp, 2017-01-11:
Ok. I understand you have to focus on the new versions. I have downloaded 1.8.5
I dont wish to update any further on this quad as its taken 6 months to get it just right, I cant keep up :-0 :-D

mikeller, 2017-01-11:
Ok. You can still use the CLI to set your values. Or upgrade to 2.9.1, which only fixes some bugs present in 2.9.0.

borisbstyle, 2017-01-11:
You can indeed also use cli. The configurator only doesnt display it correctly