betaflight/betaflight-configurator #392: two new osd fonts...

DeepnineRC, 2017-01-16:
I made two different font for the internal osd.

The first is a extralage font, i looks like the EZOsd font. I found it some years ago, i don't know where the file came from. I modified it to fit betaflight osd.

The second one is from me. I tried a "betaflight" font but is very small. I don't think its good for flying.

[image]
[btfl_extraLarge.zip](https://github.com/betaflight/betaflight-configurator/files/708661/btfl_extraLarge.zip)

[image]
[btfl_btFont.zip](https://github.com/betaflight/betaflight-configurator/files/708662/btfl_btFont.zip)

Feel free to do what ever you like with the fonts. Maybe someone would like to add them to the configurator.

mikeller, 2017-01-16:
I like the first one, it's got a very good readability.

DeepnineRC, 2017-01-16:
Thats the one i'm using for mwosd for years. :-)

Can someone add them to the configurator? I can't get the git stuff working.

DanNixon, 2017-01-28:
Added them in #410.

Love the extra large font, much easier to read when wearing goggles for me.

toddco, 2017-04-15:
Thanks, so much for the large font, better than stock. So much easier on my eyes.

closed 2018-05-13 by stale[bot]