betaflight/betaflight-configurator #405: F3 Femto: Unknown USB Device, Zadig Drivers: None

remzak, 2017-01-24:
I am trying to connect my F3 femto board to my Windows PC, I connected it once before and got the thing flying, but now after 3 weeks it will not connect to betaflight, nor will the PC recognize it at all. When I connect it I immediately get a message saying USB deice not recognized, and then in zadig it is referred to as Unknown USB Device (Device Descriptor Request Failed) and the drivers tab says (NONE) 

How can I fix it? Is it bricked?

AndersHoglund, 2017-01-24:
In boot loader mode, what does Windows Device Manager say? And Zadig? 
Still nothing, broken FC, broken cable, wrong cable, charge only cable, broken PC, broken Windows. Surely not a BF firmware issue....

borisbstyle, 2017-01-30:
Try also impulse drivers.

wretep, 2017-02-13:
I am having the same (or similar issue) with a PIKO BLX.  In boot loader mode it shows up as STM32 BOOTLOADER and I can flash successfully in DFU mode.  As soon as it's done flashing, it goes back to Unknown USB Device (Device Descriptor Request failed) and there is no virtual com port.

mikeller, 2017-02-13:
@wretep: Have you installed the VCP drivers as described in https://github.com/betaflight/betaflight/wiki/Installing-Betaflight ?

wretep, 2017-02-13:
Yes, I installed the VCP drivers, ImpulseRC driver fixer, Zadig util.. I have also tried 3 different (W10) computers and they all do the same thing..

AndersHoglund, 2017-02-14:
If you can Flash in DFU mode the the WinUSB drivers are installed correctly by either Zadig or ImpulseRC Driver Fixer. Your problem is with ordinary STM32 VCP drivers or the wrong firmware on the FC.

A common mistake is to run Zadig in non-DFU mode on the wrong USB device and destroy the VCP installation. Zadig is a bit too sharp knife, easy to cut your self. But I doubt this is the problem, as your FC does not identify itself correctly "(Device Descriptor Request failed)". The USBVCP support in firmware are responsible for this, and I would guess you have flashed the wrong target.  What target have you used? 

This FC must be flashed with the SPRACINGF3EVO target, and if you by mistake used the SPRACINGF3 target you will have exactly this problem as there is VCP support only in the EVO target.

closed 2017-07-27 by jflyper

waynewendrone, 2017-08-01:
Can I ask if you have fixed it? This just happened to my omnibus f3 pro and the gyro light is on but dimly and I get unknown USB device

jflyper, 2017-08-01:
@waynewendrone This was not a software bug that needed a fix.
Your issue look like a support request, too.
Please redirect your question to the appropriate thread on RCG.

OMNIBUS F3 Pro
https://www.rcgroups.com/forums/showthread.php?2742116-OMNIBUS-F3-Pro

waynewendrone, 2017-08-01:
I don't know either but my fc was working fine and I connected it normally before trying to flash it, during flashing it said it was unprotecting the fc and then asked me to unplug and re plug then this happened

DrUlf, 2017-10-10:
I had the same problem with an AirbotF4 board in DFU mode. It showed up as Unknown Device. Usually an "Unknown Device" indicates that the USB port does not provide enough power. (This is the reason for testing if the problem still occurs on different machines and/or different USB cables). But there are other settings that control the power a USB port delivers. In my case, the following setting helped: Energy Options -> Change advanced energy options -> USB selective suspend settings: Disable them (see image). After that the FC showed up again as STM Bootloader.

[image]

kdoubner, 2018-06-22:
Same problem with my F4. Disconnecting of GPS help.