betaflight/betaflight-configurator #421: New Font

mixMugz, 2017-02-06:
Hi. I made new font for BF. You can use if you like.

[image]

Link for download: https://cl.ly/1z1J1u02113G

DanNixon, 2017-02-06:
As this font appears to be identical to the one added in #419 I'd suggest this should just replace that one (digital.mcm`) as the battery icon does look nicer in this one.

(@basdelfos)

mixMugz, 2017-02-06:
Oh. Wow! in #419 is also my font what i make for minimOSD ;)

mixMugz, 2017-02-06:
Anyway you can rename it to anything you like and use

DanNixon, 2017-02-06:
Cool, I'll be able to get that done some time tomorrow.

mixMugz, 2017-02-06:
Okay! If you need to change something - let me know.

basdelfos, 2017-02-06:
What's the difference besides the battery level icons?
I like the battery icons from the default Betaflight font more, I left them
out on purpose. But that's just suggestive. It's possible to just upload
the font you like in the BFC.

@mixMugz I thought you wouldn't mind sharing this font as you requested it
on RCG a while ago.

mixMugz, 2017-02-06:
@basdelfos Of course i don't mind sharing ;)

I didn't notice your post. If i did i wouldn't have written.

There aren't many differences.
1. Battery icon
2. 'v' and 'a' symbol for Voltage and Current
3. Correct center cross

mixMugz, 2017-02-06:
And battery icon is better. Because it is more informative.

DanNixon, 2017-02-06:
Yeah, having the individual segments would make it easier to read at a glance IMO.
Although those icons aren't even used in the Betaflight OSD (at the  moment) so for now it makes little difference.

basdelfos, 2017-02-06:
@mixMugz I will replace it with your font, it's your font anyway :)

DanNixon, 2017-02-06:
@basdelfos If you PR it then you also need to add it to the font UI, otherwise no one will ever know it's there.

See #410.

basdelfos, 2017-02-06:
Good point, didn't notice that.

mixMugz, 2017-02-10:
And you can use original digital.mcm filename (or betaflight-digital). mugzless - temporary name.

PR #431 (2017-02-11): Renamed font 'mugzless' into 'digital' to avoid confusion.
Fixes #421.
closed 2017-02-11 by mikeller