betaflight/betaflight-configurator #426: Motors tab and BLheli suites passthrough not working when Dshot and 3d mode enabled.

Dracen-Knights, 2017-02-08:
ok so I am setting up my first Dshot quad and after many hours of test to figure out why the motors tab is not reflecting the fact I'm in 3d mode I found out that it's a glitch. I also can not get the passthrough to work when in 3d mode. now if I set it to oneshot 125 or any other non dshot the motors tab reflects 3d mode and I can use blheli suites to check the esc's . once any dshot is set then the motors tab doesn't work and passthrough doesn't work either.

My setup is as fallows

Armattan Chameleon Frame
xracer f303 v3.1 Running BF 3.1.3
Armattan 30a esc's (dshot compatible)
Armattan Oomph motors
Armattan 5 & 12v regulated PDB
Runcam owl plus camera
Eachine 600mw VTX
Furious Piggy OSD

I was really hoping to be able to use the motors tab to test the setup but at this point I can't. So how do I set deadband? or low and high points?

rotcehdnih, 2017-02-08:
Just dropping in here to sub to the issue :)

mikeller, 2017-02-11:
Ok, @Dracen-Knights, @rotcehdnih, I implemented a fix for the motors tab when 3D mode is used with DShot in #432. Unfortunately I cannot test it, since I don't have any DShot ESCs that can do 3D mode, so you will have to help me there. Can you please install configurator from my branch, and test if the motor tab works for you now?

Dracen-Knights, 2017-02-11:
Sure can just need to know where to find your branch... Lol

mikeller, 2017-02-11:
It's `mikeller:fix_dshot_3d_motor_tab`, which is what #432 is based on. :-P

To get it:
```
git checkout -b mikeller-fix_dshot_3d_motor_tab master
git pull https://github.com/mikeller/betaflight-configurator.git fix_dshot_3d_motor_tab
```

Dracen-Knights, 2017-02-12:
Well I guess I'm to much of a noob as I can't find what your talking about I don't see any files or mention of the fix_dshot_3d_motor_tab

mikeller, 2017-02-12:
You'll have to use the commands listed above to do a local checkout of the branch with my changes. See also https://github.com/betaflight/betaflight/wiki#configuration-tool on instructions on how to run a local copy of the configurator.

Dracen-Knights, 2017-02-12:
wow ok I think thats above my level right now.  I thought you did a file sorry I guess you can say I'm just scared to screw stuff up.  and as is my quad flies the way it is I will more than likely just have to wait until the betaflight file fixes the issue. I just wanted to make sure it was known of and got fixed soon.

PR #432 (2017-02-14): Fixed motor tab for DShot 3D mode.
Fixes #426.
closed 2017-02-14 by borisbstyle