betaflight/betaflight-configurator #427: Feature request: select box to reverse motor direction.

Dracen-Knights, 2017-02-08:
As I don't like sealing wires around when I build my quadcopters, (looks cleaner straight). It would be really nice to have a check box on say the motors tab by each motor that once checked would reverse that motors direction.

AndersHoglund, 2017-02-08:
Use "BLHeli Configurator" or "BLHeliSuite" for that task. Hard to do in BFC...
[image]

Dracen-Knights, 2017-02-08:
I do that now but I have also done it in chl just change the the yaw in that motor I believe it is in the mmixer.  I change the last 1 to a -1 or -1 to a 1 which evers needed.

I learned how off one of joshua bardwell YouTube videos works great.  But now all my quads are 3d, I just know I have heard it mentioned several times that a selected to reverse motor direction would be nice.

mikeller, 2017-02-09:
@Dracen-Knights: I see your point. The problem here is not a shortcoming of the Betaflight configurator (or really the Betaflight firmware), but of blheli / blheli_s:
While the input to motor direction relationship can be configured on the ESC for non-3D, for 3D it only offers '3D off' and '3D on', instead of 'off' / 'spin clockwise on input above middle' / 'spin anti-clockwise on input above middle', potentially forcing the user to switch motor wires to get the right direction.

While using `mmix` in Betaflight as a workaround for this shortcoming of blheli, I don't think it should be institutionalised. Please submit a bug report / feature request for this to blheli.

Dracen-Knights, 2017-02-09:
I see your point there I'm pretty new to all this and as a matter of fact I have only had a github log in for a few days now. Still learning how things tend to be done here.

closed 2017-02-09 by mikeller