betaflight/betaflight-configurator #428: HEADFREE mode is not highlighted

Abu-Abdullah, 2017-02-08:
Hi,

when setting HEADFREE mode to certain AUX, it is not highlighted in yellow as the rest of the modes

mikeller, 2017-02-09:
That is because the mode does not get activated. In general, modes can have additional conditions for activation (e.g. a battery has to be connected for ARM to activate), and the mode bar in configurator is only highlighted if these conditions are met.
Currently, the board has to have a MAG sensor, and either ACC or MAG sensors have to be enabled for HEADFREE to activate. This is obviously incorrect, https://github.com/betaflight/betaflight/pull/2362 implements a fix so that only either MAG or ACC have to be supported.

Abu-Abdullah, 2017-02-09:
thanks. will test the new fix once a new version is available

closed 2017-02-09 by Abu-Abdullah