betaflight/betaflight-configurator #434: Flash firmware fail for STM32F4 target

KyunghyunLee, 2017-02-13:
I am using custom board with STM32F4 target. 

I used flashing firmware via configurator and it worked well before 1.9.1

I found it is not working after "js/protocols/stm32usbdfu.js" changed. 
Before below commit, it works well. and if I overwrite only "stm32usbdfu.js" into 1.9.1, 
it works ok too. 

https://github.com/betaflight/betaflight-configurator/commit/5dbcd02148f2a01c3287273575c0c115a39c7ea8#diff-2532d20c4c383ac95f748199daeac976

mikeller, 2017-02-13:
In order to be able to help you, you would have to provide more information on how it fails. Also, please understand that we're working to provide support for boards that are commercially available, and this fix seems to work for these boards.

KyunghyunLee, 2017-02-22:
The board entered to DFU mode, then stop. 
Any sequence changed?

mikeller, 2017-02-22:
Yes. We are now trying to detect and avoid timeouts during the flash erase.