betaflight/betaflight-configurator #435: Mac OS X serial corruption/CRC errors investigation

etracer65, 2017-02-13:
Although a workaround was developed to deal with serial corruption in MSP packets in BF 3.1+, I wanted to continue the investigation of the underlying cause.

**Here are some observations:**

1. Only seems to be affecting Mac OS X clients

2. MSP packets from the flight controller to the Configurator are being corrupted and are triggering CRC errors.  Analysis of the packets seems to indicate that the next message is beginning before the current message has been completely received as the preamble "$M>" is found in the payload.

3. The corruption appears random, however it's common for there to be corruption at the very end of the blackbox download.  Also the CRC errors are followed by 1 or 2 timeouts of other MSP messages (probably because they were "consumed" in the previous message with the CRC error).

4. The implementation of jumbo packets for blackbox download appears to made the problem worse.  I suspect this is due to the larger transmissions triggering or exacerbating the underlying problem.

5. Baud rate doesn't seem to matter.

6. Reducing the jumbo packet size doesn't seem to help unless really small values (<256 bytes) are selected.

7. Not every user is affected.  Faster machines appear to have fewer or no occurrences.  For example I have a high-end MacPro which cannot replicate the problem where an older MacBook Pro and iMac can.

8. Occurs with both CP210x and VCP targets.

9. Can't find any pattern on OS X version.  Tested 10.8 through 10.12 and can replicate.

10. No correlation to USB cables, hubs, etc.  Tested many combinations using different computers.

**Items to investigate:**

1. OS X drivers.  Since the problem can be replicated for both SiLabs CP210x and STM VCP drivers it's unlikely to be a driver problem.  Additionally I tested some very old versions of the drivers and this also made no difference.  Lastly, unable to find any widespread reports of serial problems with either vendors drivers.

2. Possible problem with the FC firmware serial implementation. Possibly a configuration or implementation problem causing data to be lost in transmission (flow-control, etc.).  Seems unlikely as that would affect all platforms.

3. Generalized problem with OS X USB implementations.  No widespread reports and personal experience with other high-speed USB/serial interfaces would dispute this.

4. Serial handling or other bug introduced in the OS X version of Chrome.  Couldn't find any related reports but it wouldn't be the first time a Chrome bug was introduced that impacted the Configurator.  However I tested a number of old versions (all the way back to version 49) and could still replicate the problem.  So it would seem unlikely to be a recently introduced Chrome bug.

5. Bug in the Configurator.  Possibly there is some flaw in the Configurator serial/MSP message logic.  Again it seems unlikely as you'd expect this to affect all platforms.

**My investigation so far:**

I wanted to see how many of the possibilities I could eliminate so I decided to **write my own standalone blackbox downloader outside of Chrome**.  If this could successfully and repeatedly download the logs then it would eliminate possibilities 1-3.

So after more work than I originally planned :) I was successful in creating a standalone download program.  With this tool I'm able to successfully download logs at high baud rates with absolutely no CRC errors on machines that repeatedly fail using the Configurator (and same flight controllers).  So in my mind this invalidates possibilities 1-3.

This still leaves possibilities 4 and 5 open.  So far there's no evidence of a serial communications bug in Chrome.  I supposed a test platform could be constructed as a Chrome app that stress tests the serial handling.  To replicate the environment (MCU, drivers, etc.) it would probably also require a custom firmware that sent large amounts of data.  This is beyond my skills at the moment so I'm continuing the investigation of the Configurator serial/MSP message handling.

One other thing I investigated was how the serial port was configured/opened in the Configurator.  The default buffer size in he Chrome API is 4096 bytes.  Since is also the packet size used in jumbo frames I was concerned this might contribute to overrun buffers, etc. so I tested with an increased bufferSize but this unfortunately made no difference - scratch this off the list.

etracer65, 2017-02-13:
So even though I was successful in creating a standalone blackbox download application outside of Chrome, it turned out to be much more work than originally planned.

My goal was to completely isolate the environment from Chrome and used Xojo - don't worry if you don't know what this is - it's a cross-platform IDE that I'm familiar with.  Not really relevant except that it's not related to Chrome or even a web-based application.

My next goal was to replicate the architecture used to receive and process the MSP messages in the Configurator.  This includes an "onReceive" event on the serial object with a callback that processes the received data.  The data is processed through a state engine that decodes the message, captures the payload and verifies the CRC of the message against the contents, and then finally executes a callback to service the message payload.

So when I first built my application this way I found that I also received apparently corrupted data that generated CRC errors.  Good… I guess.  This basically meant I wasn't able to discount any of the possibilities.

However, since I was now on a platform that I was more comfortable with it was easier for me to debug and understand what was going on.  And since I was able to successfully get the application to work after re-architecting, I think the underlying cause that I discovered might be applicable to what's going on with the Configurator.

In a nutshell, the problem was being caused by having the logic to process the message directly in the callback from the onReceive event.  With the addition of the jumbo packets there is now over an order of magnitude more data to process in the callback then previously.  What I observed in my application was the onReceive event being fired again due to new data **while the previous block was still processing**.  Since the state machine control data is persistent to allow partial receipt of messages, the second instance of the onReceive callback with new data would begin to process and corrupt the state machine.

To resolve the problem I had to redesign the process to have the onReceive callback simply buffer the newly received data (I used a ring buffer) and let a separately scheduled service routine actually process the buffer.  This isolated the lengthly processing from the onReceive event and once I did this all the problems were solved.

I think this might be what's going on in the Configurator as it addresses a number of the observations:

1. Problem is worse with larger packet sizes - makes the onReceive callback take longer with more risk of a collision with the next callback.

2. Not occurring on faster computers - able to process the buffer quicker and less likely to collide.

3. Finding the preamble "$M>" in the corrupted messages. Indicates the second callback starting to process before the previous completed.

4. Frequent "last download packet crc error" - The last packet of the blackbox download is usually a partial block size - this would send quicker and fire the onReceive sooner.

Now this doesn't explain why it's only happening on Mac OS X and not other platforms.  Possibly there is a difference in the Chrome implementations of event handling due to underlying OS concepts.  For example, I'm not sure if Chrome events are multi-threaded (run in parallel) or interrupt/return model.  However either model would cause corruption of the state machine in the current Configurator architecture.  Ideally the subsequent onReceive event should be delayed until the previous completes.  Maybe that's what's supposed to be happening by isn't.

mikeller, 2017-02-13:
Thanks for looking into this, @etracer65.

odixxx, 2017-02-21:
the problem appeared on windows
Version 58.0.3013.3 (Official assembly), dev (64 bit)

etracer65, 2017-02-21:
@odixxx Do you have a link you can share?  I think I've eliminated every possibility but a Chrome bug on the OS X side so maybe it's the same issue.

odixxx, 2017-02-21:
where to watch an error here or elsewhere
[image]

odixxx, 2017-02-22:
the next time the application restarts another error
[image]

etracer65, 2017-02-22:
@odixxx I'm not sure what you're trying to show or if this is the same issue.  This investigation is about dropped or corrupted serial data leading to CRC errors in the MSP message handling in the configurator.  So far this has only been reported on Mac OS X implementations in Chrome.

fiam, 2017-09-20:
@etracer65 We're suffering this problem on INAV too and I'd like to check the workaround you mention that went on BF 3.1, since it might help us too. I've searching both the betaflight/betaflight and betaflight/betaflight-configurator repos by I can't find it. I've looked at all issues PRs that mention CRC or corruption or macOS, but no luck. Would you mind linking the relevant PR/commit? Thanks!

etracer65, 2017-09-20:
@fiam  We were never able to find/fix the cause of the serial corruption so a workaround was put in place to retry any blackbox download packets that contained CRC errors.

See https://github.com/betaflight/betaflight-configurator/pull/424

Note that many additions have been made to blackbox downloading since including jumbo packet sizes, data compression and other optimizations.  But the retry logic is still active and needed to prevent stalls due to CRC errors.

fiam, 2017-09-20:
@etracer65 Thank you so much for digging that up for me! Just a heads up in case it ends up hitting ßF too: we're also seeing the problem when sending exactly 64 byte messages from the host to the FC. It happens when uploading a font via MSPv2, since each character is 55 bytes and MSPv9 adds another 9 bytes on top.

Strangest thing is that I can reproduce it on an OMNIBUS (F3) but not on an OMNIBUSF4PRO (F4). Uploading a font from Windows on that same F3 board works without problems, so it's definitely macOS specific.

This is the issue with our findings: https://github.com/iNavFlight/inav/issues/2054

closed 2018-05-06 by etracer65