betaflight/betaflight-configurator #436: Feature Suggestion - Camera weight Compensation

quadmx, 2017-02-13:
What it is about?
Make an offset on the throttle to compensate for the added weight of a camera/or other component (such as a havier battery)

Backstory
People struggle a bit when having the extra weight(gopro,bigger battery,etc) since throttle requires more input.
Thus I would like to suggest that an option can be added for camera weight compensation in which your throttle curve is offset upwards in order to cope with the different weight (on PID page, you could enable/dissable per profile).
Since all drones have different weights and motors, etc. Maybe it can be something similar to your vbat option, but here 0 is no offset and + numbers offset the curve.
For tunning I would basically be by trial and error. You increase some steps and test to see if it is behaving as if it had no extra weight. (Maybe add the option to have tunning possibility through the aux switches).
When throttle is zero or near zero then offset disappears (so that airmode works similar and you don't have that extra throttle when doing flips to push you to the ground.)
Regards and thanks for the awesome job the team is doing

mikeller, 2017-02-13:
This is not something that can be done in configurator, since the throttle curve has to be applied by the firmware, and not by the configurator.

Also, there is a redo of the throttle curve configuration planned for 3.2. This will allow more customised throttle curves, allowing people to configure for added weight, amongst other things.

closed 2017-02-13 by mikeller

quadmx, 2017-02-13:
Thanks a lot!