betaflight/betaflight-configurator #439: [Feature Request] endpoint/midpoint padding

Viss, 2017-02-17:
sp racing f3 acro FC
Was previously running cleanflight. 
Flashed to betaflight, midpoint on yaw went from 1500 to 1435 or so (now quad yaws left)
Midpoint on roll went to 1520 or so (add roll right to yaw left)

I asked for help in the rotor riot facebook group (that was a mistake).
I'd really like to be able to trim out the midpoints to the correct figures from the fc rather than adjust my radio (tbs tango) for every single aircraft, and have to tune the radio AND the quad separately.

it would be really really amazing to be able to pad the values in the fc to do the trim there. 

I hope what I'm asking for is not an insane ask, I honestly have no idea.

mikeller, 2017-02-18:
Hi @Viss.

Are you saying that all of your channels have different midpoints now? That is very odd. Betaflight allows adjustment of the RC midpoint for ***all*** channels, but not for individual channels.

How is your RX connected to your flight controller?

le3dmax, 2017-02-20:
Does the TBS tango have sub trim?

*LD Silva.*

https://www.instagram.com/ldsilva_fpv/

https://www.facebook.com/fpvldsilva/
https://www.facebook.com/ldsilvafpv/

https://www.youtube.com/user/le3dmax/

Viss, 2017-02-20:
I dont think so. 
Only the main physical trim on the tx. 

The whole point here though is 'it would be really nice to not have to manually reconfigure the tx for every single aircraft,  but rather just fix the trim in the fc,  and never deal with the tx trim again'

mikeller, 2017-02-21:
@Viss: You can do that. Just set 'Center value for RC channels' on the Receiver tab.

Viss, 2017-02-22:
@mikeller isn't that one value for all channels, not just one channel? The problem i'm experiencing is only on one channel, so this would 'fix' the yaw issue, but it would introduce a trim issue on every other channel.

mikeller, 2017-02-23:
@Viss: It sounds highly unusual for a TX/RX to have different midpoints with default settings, so I don't think that making the channels individually adjustable makes much sense. Are you sure that this is not caused by trims on your TX?

le3dmax, 2017-02-23:
Are you using the latest tbs tango firmware? My friend fixed it after the
update.

*LD Silva.*

https://www.instagram.com/ldsilva_fpv/

https://www.facebook.com/fpvldsilva/
https://www.facebook.com/ldsilvafpv/

https://www.youtube.com/user/le3dmax/

closed 2017-03-14 by mikeller

NinjaSauce1, 2018-05-29:
Ok. So Im looking for some similar info. Basically, Im wondering if there is a way to change a single aux channel midpoint. OR is there a way to change an aux channel activation range point? (Not the range, but the point where that channel activates) Im having a problem where turtle mode is active at 1500. And without the transmitter powered on, that channel rests at 1500, so turtle mode is active. Its really not a huge problem. I was just wondering if there is a way around this. Im getting sick of the beeper going crazy when Im working on it.