betaflight/betaflight-configurator #440: Configurator extremely slow reaction

Xguy4ku, 2017-02-18:
Betaflgiht started behaving strangely about a couple of months ago, first the 3D model disappeared, so I blamed it on my graphic card updated drivers bla bla nothing seemed to help, I have a fairly fast computer with dual graphic radeon, running windows 10, quadcore processor 12 gigs of ram, so nothing should be lacking there.
Now I got the 3 model fine but it takes about 5 seconds to react to my moving the quad/board, that's not really a big deal but it's not normal, but the second issue wich is very strange is that when I changed PID's and hit save I exit and come back and PIDs are back to previous values, at first I thought they were not saving at all but then I watched it as I hot save and it took about 30 seconds yes 30 that's not a typo to get the message on top saying saved to eepron, the faster way of doing it was to go from pids tab into the configuration tab and save and reboot, it will then save PID's.
things I've done uninstalled and reinstalled betaflight, installed the latest zadig driver tried to run betaflight in different windows compatiblities, nothing helps, 
thanks

mikeller, 2017-02-19:
>  the faster way of doing it was to go from pids tab into the configuration tab and save and reboot, it will then save PID's.

What you are describing is a way to *not* save your PIDs, since the values will be discarded and reloaded from the flight controller when switching tabs.

If your configurator is running slow, it's probably because your Chrome is outdated or broken. Try uninstalling Chrome, and then reinstalling.

Xguy4ku, 2017-02-19:
Thank you Mikeller, although I didn't do exactly do what you prescribed, but your post led me into the right direction so I did some chrome clean up I also found that flash was disabled in chrome, after some tuning and fiddling around with chrome, betaflight now works fine, 3D model is running very smooth .

closed 2017-02-19 by Xguy4ku