betaflight/betaflight-configurator #444: Configurator is only detecting 128kb flash for an STM32F722

MJ666, 2017-02-22:
Configurator is only detecting 128kb flash for an STM32F722 (has 512kb) and refuse to flash the firmware since its larger than this. Have no idea how to debug this?

mikeller, 2017-02-22:
F7 support will probably have to be added to https://github.com/betaflight/betaflight-configurator/blob/master/js/protocols/stm32usbdfu.js#L299 and https://github.com/betaflight/betaflight-configurator/blob/master/js/protocols/stm32.js#L260 .

MJ666, 2017-02-23:
With the STM32F745 there is no problem and flashing is working without issue.

blckmn, 2017-02-23:
Is it the board identifier not being recognized in the Configurator?

mikeller, 2017-02-23:
@blckmn: Either that, or in the case of DFU being used, it could be that the parser for the memory descriptor is failing for whatever is returned by the STM32F745, which is something I can't test since i don't have a board with STM32F745.

MJ666, 2017-02-23:
My STM32F722 does not have a firmware yet. So no board ID but in dfu mode. Flashing is working for an F745 as usual for other MCU's.

mikeller, 2017-02-23:
Can you debug the configurator at  https://github.com/betaflight/betaflight-configurator/blob/master/js/protocols/stm32usbdfu.js#L299, and see what is returned by the MCU, and how it's parsed?

MJ666, 2017-02-23:
You have any info how debug is working for chrome apps? Never done this before. Thanks.

MJ666, 2017-02-23:
Found how to debug. Here is the result string "@Internal Flash  /0x08000000/04*016Kg,01*64Kg,03*128Kg". Will see how it is parsed. looks to be not much different to the F4?

MJ666, 2017-02-23:
The STM32F722 has some trailing non ASCII data in the string and this will make parsing of the last flash sector range fail. Also some of the other strings provided by this MCU have some non character bytes in between. I have pushed an related update #480 and flashing is working now properly.

mikeller, 2017-02-23:
Welcome to chrome app development! ;-) And thanks for the fix.

MJ666, 2017-02-23:
Was a nice experience and more easy than expected. ;-) I had already another fix fro the configurator quite while ago but this was fixed without the need to debug.

closed 2017-02-23 by MJ666

mikeller, 2017-02-23:
The Chorme inspector is actually not too bad as a debugger.