betaflight/betaflight-configurator #445: Not possible to select other serial rx provider than SPEKTRUM1024 in 1.9.2.

AndersHoglund, 2017-02-23:
Selecting any serial rx provider other than SPEKTRUM1024 is impossible. Whatever you select and save, it always falls back to SPEKTRUM1024. Only with latest 1.9.2 version, works OK with 1.9.1. 
 
Ref: https://github.com/betaflight/betaflight/issues/2471

Reinsettle, 2017-02-23:
I'm having the same problem. I don't know how to revert back to 1.9.1

mikeller, 2017-02-23:
Hang in there guys, we are looking into this and will be pushing a fix shortly.

closed 2017-02-23 by borisbstyle

bakerman1231, 2018-02-23:
same  problem, i have a microlux f4 and it stuck and i had it working with sbus. i went back to use the motor stop function because i dont like my motors spinning as soon as its armed. been nothing but spektrum1024 since. ive tried every allowable option in the ports starting with #6 as dictated by the microlux manual.

PR #447 (2017-02-23): Fixed setting of serialRx provider.
Fixes https://github.com/betaflight/betaflight/issues/2471.
Fixes #445.