betaflight/betaflight-configurator #448: Accelerometer graph doesn't work since 1.9.2 in motors tab

jurkust, 2017-02-23:
Accelerometer graph doesn't work in motors tab.

jurkust, 2017-02-23:
It looks like this dispite if accelerometer is on or off:
[image]

Another question is why accelerometer is used instead of gyroscope in motors tab? Most (if not all) acro pilots do not use accelerometer at all and just gyroscope is used in pid loop. So, shouldn't be gyroscope used to measure vibrations when testing motors?

mikeller, 2017-02-23:
Sorry, my bad, I was a bit too eager in cleaning up dead code.

But it's a good point that you are making about gyroscope data being a better graph to display on the motors tab. Will change this in a future PR.

PR #451 (2017-02-23): Fixed accelerometer graph on motors tab.
Fixes #448.
closed 2017-02-23 by mikeller