betaflight/betaflight-configurator #449: Suspicious permissions requirements in new version of betaflight-configurator.

janekbaraniewski, 2017-02-23:
Today when I've opened the BF-configurator this showed up:

[image]

Everything sounds reasonable apart from the first point. I honestly don't believe that in order for application to work it needs permission to exchange data using the network with any other device whatsoever. The only devices it should exchange data with are accessed via USB. Mind explaining this one?

mikeller, 2017-02-23:
Sure. This is required for the functionality to communicate with a flight controller over WiFi and an ESP8266 running esp-link, as discussed in #425. But well spotted, we should probably have placed a note about this somewhere.

janekbaraniewski, 2017-02-23:
All clear! Thanks for the quick response.

closed 2017-02-23 by janekbaraniewski