betaflight/betaflight-configurator #461: Backup feature not working

SupahDave, 2017-03-07:
I noticed that BetaFlight was updated to 1.9.3 recently and now my BACKUP button doesn't do anything when I press it. The RESTORE button works fine but I am now unable to backup my configurations other than manually doing a CLI dump backup. Any ideas as to what would cause this? 

I've tested this with my Eachine QX95 and also my Diatone Tyrant S and I can't get the feature to work. Please advise.

mikeller, 2017-03-07:
Hmm, that's odd. There is a bug in restore in 1.9.3 (fixed in #460), but backup should work. Can you open the developer console (right click, 'Inspect'), and check if you get an error message there when you click 'Backup'?

SupahDave, 2017-03-07:
I don't understand. Where am I supposed to right click?

SupahDave, 2017-03-07:
If you were telling me to right click in the CLI window, it doesn't do anything. I've tried right clicking everywhere in the interface and nothing happens

mikeller, 2017-03-08:
That is odd. When you go into the Chrome menu / More tools, does it show you a keyboard shortcut for the 'Developer tools'? Try that one when in the configurator.

SupahDave, 2017-03-08:
> <betaflight-configurator@noreply.github.com>
> working (#461)

Okay, I can get into that menu. This is what it shows when I open it 
with the configurator open... Right clicking in the developer tools 
window does nothing.

SupahDave, 2017-03-08:
[
[image]
](url)

mikeller, 2017-03-08:
That's the one. now click on 'console' at the top, and then click on the 'Backup' button in the configurator, and see if you get an error message.

SupahDave, 2017-03-08:
I tried that and nothing shows up in the console window at all when I press backup in the configurator... What does that mean? I've already uninstalled and reinstalled the configurator in case something was corrupt. What should I try next?

SupahDave, 2017-03-08:
[image]

mikeller, 2017-03-08:
Ah. This is the console for the web page you're currently browsing. You need to open it in / for the configurator window. Can't help you how to do this, if your right - click menu does not show up.

SupahDave, 2017-03-08:
So I should be able to right click in the Betaflight window? From CLI?

mikeller, 2017-03-08:
From anywhere in the configurator really.

SupahDave, 2017-03-08:
I just setup Betaflight on my laptop and tried right clicking in the window and it doesn't do anything on the laptop either. This eliminates any possibility of it being my PC that is causing the issue. Could this be an incompatibility with Windows 10?

mikeller, 2017-03-08:
Shouldn't be, works fine for me:

[image]

You don't use your mouse left handed by any chance, and have the mouse buttons swapped?

SupahDave, 2017-03-08:
Nope

SupahDave, 2017-03-08:
Does Betaflight depend on any plugins in Chrome that may be getting blocked?

bilson, 2017-03-08:
@SupahDave 

- Clone the repo to any local directory or download it as zip
- Start Chromium or Google Chrome and go to tools -> extension
- Check the "Developer mode" checkbox
- Click on load unpacked extension and point it to the Betaflight Configurator directory (for example D:/betaflight-configurator)

mikeller, 2017-03-08:
@SupahDave: Sorry, my bad. If you installed the configurator from the web store, the process to get a console is different:
- start configurator;
- enter `chrome://extensions/` into chrome's url bar;
- on the page that opens, find configurator, click on `main.html` under 'Inspect views'.

SupahDave, 2017-03-09:
Can you give me a screenshot of what you mean? I know how to find the Chrome Extensions page easily but I do not know where to find the "main.html" under "inspect views"

SupahDave, 2017-03-09:
[image]

SupahDave, 2017-03-09:
Nevermind. I found it. I had to enable Developer mode in the extensions page

SupahDave, 2017-03-09:
This is what I get when I press the backup button in the configurator

SupahDave, 2017-03-09:
[image]

SupahDave, 2017-03-09:
[image]

SupahDave, 2017-03-09:
[image]

SupahDave, 2017-03-09:
[image]

SupahDave, 2017-03-09:
I upped screenshots of what the console shows when I click on each of the links

mikeller, 2017-03-09:
From looking at your screenshots, the 'error' seems to be 'User cancelled', i.e. Chrome thinks that you hit 'Cancel' in the dialog that asked you to specify where you want to save your backup file. If this happens without you actually hitting cancel, then I suspect something is wrong with your Chrome installation.

I'd try uninstalling and reinstalling Chrome.

SupahDave, 2017-03-09:
I uninstalled Chrome from my laptop and reinstalled it without signing in to Google so that it wouldn't sync anything from my PC's browser. I then installed Betaflight from and tried it again and I get the same error that I already posted pics of (those pics were from my PC). 

So both my PC and laptop are experiencing the same issue even with a fresh install of Chrome/Betaflight on my laptop... 

Any other suggestions?

tyler78, 2017-03-09:
I have the same problem with "backup".  There's also post over in the betaflight by woody37 that also sounds like the same problem.... So not isolated to just SupahDave....  happy to do some troubleshooting if there are additional recommendations.

SupahDave, 2017-03-09:
Thanks for the validation Tyler. Nothing has changed on either of my computers other than the Betaflight version being updated automatically from the Chrome app store. I wish there was a way to downgrade it to a version from January because I know that was working then...

mikeller, 2017-03-09:
Ok. I think I can see what goes wrong. Looks like Chrome auto-closes the file selection dialog if the suggested file name contains characters that cannot be part of a filename, which is the case with the currently released version under windows. Unfortunately it does so and logs this with the utterly unhelpful message 'User cancelled'.

I will get a new version released.

SupahDave, 2017-03-09:
Excellent! Glad you were able to figure this out since it was apparently happening to several people.

SupahDave, 2017-03-10:
Great job diagnosing and fixing the issue. Backup is working again with the version of betaflight that is on the Chrome web store now. Thanks!!!

closed 2017-03-11 by mikeller

ChristianProgrammerTaylor, 2018-02-26:
excuse noob observation but I believe this is happening ( albiet not a browser version ) in the new 10.1  not chrome  configurator