betaflight/betaflight-configurator #466: [Feature request] BLE (HM10/HM11) support

CoreyOsgood, 2017-03-12:
Would it be possible to get support for connecting to the flight controller over BLE? EZ-GUI works OK on Android, but it would be nice to be able to use the full Betaflight configurator over bluetooth when using an HM10 or HM11 module.

Mistale, 2017-04-16:
I second that. I've just ordered 5 HM17's to put in my quads.
Interestingly, html5 supports bluetooth-communication as well (draft, available in latest chrome + chrome android).
That would perhaps mean that android users could access the configurator as a webpage and connect via BLE.

Furthermore, the configurator should be able to be wrapped via cordova or similar, and provide access via BLE for both ios and android.

If no interest exists for this feature, I'll probably explore it myself once my modules arrive.

fbohmann, 2017-04-16:
when your computer is paired with the bluetooth module it will have a bt-serial port you can use with configurator like any other serial port. i am using this with a hc06.

CoreyOsgood, 2017-04-16:
Not with BLE (Bluetooth Low Energy), the device doesn't have the virtual
serial ports like EDR serial devices do.

mikeller, 2017-05-29:
How does the BLE module on the craft connect to the flight controller?

CoreyOsgood, 2017-05-29:
It's a UART (serial) connection to the flight controller. These are the
modules I'm using, they're much smaller and lighter than the hc-05/hc-06,
and have lower power consumption.

http://wiki.seeed.cc/Bluetooth_V4.0_HM_11_BLE_Module/

The HM-10 is the same size as an HC-05/HC-06 and available with a similar
breakout board, but basically the same components as the HM-10. HM-16 and
HM17 are newer Bluetooth 4.1 versions of the HM-10 and HM-11.

I don't know if this helps, but EZGUI on Android already supports
connecting to the FC via BLE.

matoy, 2017-06-21:
Hi,

I think that too!
That would be a nice feature to be able to select a paired BLE device in the COM ports listbox when connecting.

Thank you for your hard work.

truk14, 2017-07-11:
Count me in as looking for this. BLE modules are good for the brushed micros.

tloki, 2018-06-04:
any news about this?

marianomd, 2019-10-05:
This would be great. It would avoid usb connector wear. This chrome example detects my HC-10 and connects to it succesfully: [GitHub.io](https://googlechrome.github.io/samples/web-bluetooth/device-info.html)

dharmik768, 2024-04-03:
It would be great to have a configurator over BLE for communication with ESP32 or similar development boards. SpeedyBee accomplishes this task effectively.

apk015, 2024-06-11:
**Subject: Help Needed for Wireless Bluetooth Connection with Betaflight/INAV**

Hi everyone,

I need help setting up a wireless Bluetooth connection for Betaflight and INAV using a SpeedyBee F405 flight controller with built-in Bluetooth. I've got the web Betaflight configurator working and can connect, but I need advice on maintaining a stable connection and integrating this with Betaflight and INAV configurators. Any tips or resources would be greatly appreciated!

Thanks,
SURAJ ( FLYSPARK )

CoreyOsgood, 2024-06-11:
This is not the place for support.

closed 2024-06-28 by haslinghuis