betaflight/betaflight-configurator #468: Motors tab gyro graph

killerm1, 2017-03-13:
Hi,
 I use the motors tab acc graph to balance motors and since it has been changed to a gyro graph all I see is erratic behavior on the graph. I have tried it with 3 different FC's and they all do the same thing. Is it normal for  the gyro to jump all over the place when sitting level on the desk? Is it possible to change it back to the acc graph or maybe have a selector to choose which graph you want to see?
thanks

mikeller, 2017-03-13:
Have you tried reducing the sensitivity in the 'Scale:' drop down in the right?

killerm1, 2017-03-13:
from what I can see that doesnt reduce sensitivity just changes the scale of the graph. also I have noticed the reset button doesnt work.

mikeller, 2017-03-13:
The reset button was fixed in #464.

HerrTim, 2017-03-14:
Do I have to update the file manually or how does it work?
I'm having the same issues than killerm1. Is it possible to change back to the Accelerator?

mikeller, 2017-03-14:
You don't have to do anything, iIt will be included in the next update of the Betaflight configurator.

HerrTim, 2017-03-16:
Thank you. Do you know why the Accelerator graph has been replaced?

mikeller, 2017-03-16:
Because it was pointed out that Betaflight defaults to having the accelerometer graph disabled, making the accelerometer graph useless out of the box (#448).

I am neither here nor there, I will probably make it user selectable when I get around to it.

csbuzkp, 2017-05-31:
Could you tell me when there will be an update fixed reset button?

mikeller, 2017-05-31:
@csbuzkp: It's fixed in current master (#464), which will be released together with the firmware 3.2 release.

OtterNas3, 2017-08-20:
Is this still going to happen for a newer(ish) version of the BF Configurator?

mikeller, 2017-08-20:
@OtterNas3: Definitely. It's just waiting for somebody to implement it...

KoDaKrom, 2017-10-18:
Somebody could explain to me if it's normal that the gyroscope realtime values are not stable and fluctuate even the quadcopter is on a table without any mouvements ???

In these conditions, if you go to sensors tab, you ca see that accelerometer value are stable. Gyro values "moves" (sorry for my english...) equaly to the motors tab.

I would use the motors tab to verify the balancing of my motors, but actually it's not very usable with the gyro values/graph.
Reset button works on 3.2.1 release but it's no sense to use it. xyz value change all the time!

@OtterNas3: I'have the same request as you, having an option to choose Acc or Gyro would be great.

OtterNas3, 2017-10-19:
@KoDaKrom 
Download the new Betaflight configurator from the Github here:

closed 2018-05-03 by mikeller