betaflight/betaflight-configurator #470: Would it be possible to redefine the RC curve?

nateimig18, 2017-03-15:
As in one that focuses on taking parameters like "center-stick" slope; 'max-stick' end points; & "center-linearity" width? I've been making matlab scripts for both Kiss & Betaflight that find the range of curves based on these first two parameters, and noticed theres (respectively) room for improvement.

I developed a curve that always keeps these first two parameters constant while varying this "center-linearity" width on much broader range. I'll attach the derivation document and a link to the project with the scripts:

https://github.com/nateimig18/AcroRcCurveFinder

[rcCurve_Redefine.pdf](https://github.com/betaflight/betaflight-configurator/files/846103/rcCurve_Redefine.pdf)

borisbstyle, 2017-03-22:
I think this should belong to betaflight fc and not configurator?

nateimig18, 2017-03-28:
Yea your right, found this out later after posting (assuming it was interpolated from a lookup table). https://github.com/betaflight/betaflight/blob/master/src/main/fc/fc_rc.c#L99

I now have efficient optimized c-code for the powf() function waiting approval? As well as two different versions of redefined rc curve functions. Both have the same gamma portion (x^g), but the first is added with a simple linear term & the second is added to an exponential function (with an additional parameter).
[image]

Would it be more appropriate to talk on rcgroups by starting a new thread somewhere?

borisbstyle, 2017-03-28:
Well this is the right place to talk. I was just saying that it doesn't belong to configurator section.

If you want apply a PR with your changes to betaflight for discussion?

nateimig18, 2017-03-29:
Sorry for the delay!

I'm a little hesitant, being a noob to GIT right now and not having a proper tool chain setup to debug any changes i make (just a tad embarrassing). I uploaded the C-code to another GIT repo that works with Codeblocks & Koolplot to visually verify the approximations as well as the outputs of both RC curves.
https://github.com/nateimig18/HypeRCurve

Kinda was also wondering your thoughts or concerns, if its something you think could be added?
I mean this whole idea started with reverse engineering each of BF & KISS RC curve's, to control intuitive parameter's such as center-stick slope, & max-rate, noticing there's a very limited range of curves that meet this criteria, I wanted to try and find a better one that's more intuitive [at least to me atm].

closed 2018-05-13 by stale[bot]