betaflight/betaflight-configurator #480: Implementation of Configurator RC Curve not the same as the FC RC Curve.

nateimig18, 2017-04-07:
Looks like the RC Curve inside the configurator code does not match the same implementation in the FC code. Specifically the `rcInput` in the configurator is cubicly remapped where as the `rcCommandfAbs` is not.

[image]

[image]

borisbstyle, 2017-04-07:
Yeah.....it seems one uses 3 power exponential and configurator 2.

I think its my mistake on the configurator side.
But feel free to experiment with different curves

borisbstyle, 2017-04-12:
Actually just checked and it looks all same to me

nateimig18, 2017-04-13:
So I was looking at the current master source for both FC and Configurator: 
https://github.com/betaflight/betaflight/blob/master/src/main/fc/fc_rc.c#L174
https://github.com/betaflight/betaflight-configurator/blob/master/js/RateCurve.js#L83

**Betaflight FC Implementation**
There are two variables initially used here `rcCommandf` & `rcCommandfAbs`
[image]

**Betaflight Configurator Implementation**
Here though there initially is only one variable `inputValue` and then later the cubic remapping is stored back into `inputValue` before a second variable `rcInput` is declared and initialized with `inputValue/maxRC`.   
[image]

closed 2018-05-13 by stale[bot]