betaflight/betaflight-configurator #486: Battery Voltage / Current Sensor configuration needs update

MJ666, 2017-04-26:
Since feature VBAT and CURRENT_METER are removed from the firmware some updates at the Configuration tab are needed. The sliders for the features need to be removed for newer firmware. All related fields for this two settings don't work properly at the moment with latest master. the don't show the proper settings and updating them is also not working. Likely the MSP has changed to much.

mikeller, 2017-04-27:
I think @hydra has fixed this in cleanflight-configurator (or is at least in the process of fixing it), and changes should be pulled over from there, together with the other configurator changes that are required for the new battery code.

hydra, 2017-04-27:
it's partially fixed in cf configurator, just pull the changes for the api 1.33.0 branch.  when i say partially fixed it means that the power tab is working with the new api apart from being able to change the voltage/current meter settings and send them to the fc.  the other battery settings are working fine however.

basdelfos, 2017-07-16:
This issue has been finished with PR: https://github.com/betaflight/betaflight-configurator/pull/547

closed 2018-05-06 by MJ666