betaflight/betaflight-configurator #511: Confusing RSSI channel listing

MathijsG, 2017-05-30:
While having trouble setting up RSSI and finally got it working later, I discovered that the RSSI dropdown channel list just uses the regular channel numbers, which are different from the AUX numbers. This was the first time I saw the raw channels listed. So when you are mapping functions and flight modes and stuff to AUX channels all the time, it could be quite confusing to pick the right channel out of a raw channel list

And because I'm not the only one I want to suggest some clarification. Here a screenshot of how it looks currently:
[image]

Maybe we can make the RSSI channel list dropdown somewhat more clear, maybe also output their current function, the same as in the left list. Now people easily mistake AUX 5 with channel 5, while you need to add the 4 control channels and pick channel 9 if you wanted to choose AUX 5. Or maybe add the raw channel number in the regular list on the left. I always wondered why my channel 5 on my receiver wasn't AUX 5 in my flight controller. But now I understand it all, but I'm opting for a consistent notation of channels.

See for my initial issue: https://github.com/betaflight/betaflight/issues/3181

mikeller, 2017-05-30:
You are right, this is not entirely clear.

My suggestion would be to use the channel names in the drop down.

MathijsG, 2017-05-30:
Are there more places in the configurator where you need to fill in raw channel numbers? Or is the AUX notation the main form of presenting the channels? I'm with you on the solution.

piepluquad, 2017-08-20:
from receiver.js, maybe this could be sufficient ? :  
```
for (var i = 1; i < RC.active_channels + 1; i++) {
            rssi_channel_e.append('<option value="' + i + '">' + i + '</option>');
```
becomes : 
```
 for (var i = 5; i < RC.active_channels + 1; i++) {
            rssi_channel_e.append('<option value="' + i + '"> AUX' + i-4 + '</option>');
```
so that : 
 * get rid of channels 1->4 that are reserved for Roll Pitch Yaw & Throttle
 * displays start from "AUX 1" (with inner value unchanged of 5)

mikeller, 2017-08-20:
@piepluquad: Yep, that looks good. Do you want to submit a PR for this?

piepluquad, 2017-08-20:
looking at receiver.js, aux names are found with chrome.i18n.getMessage(), so better should be : 

```
 for (var i = 5; i < RC.active_channels + 1; i++) {
            rssi_channel_e.append('<option value="' + i + '"> ' + chrome.i18n.getMessage("controlAxisAux" + (i-4)) + '</option>');
```

closed 2017-08-21 by mikeller

piepluquad, 2017-08-21:
You know mike, i didn't even tested it... is there a way to test modifications to betaflight-configurator after having done a change in my repo ?

mikeller, 2017-08-21:
Yes there is, follow https://github.com/betaflight/betaflight-configurator#alternative-way to make Chrome use the version in your local repository.

piepluquad, 2017-08-21:
ok, i tested this one, and seems ok on channel list for rssi

commit 06962fdd (2017-08-21): Merge pull request #626 from piepluquad/master

Confusing RSSI channel listing (fixes #511)
commit 59605ad4 (2017-08-21): Confusing RSSI channel listing (fixes #511)