betaflight/betaflight-configurator #515: Blackbox not reporting mode changes correctly

rodgerramjet, 2017-06-06:
Version: 2.5.11
BF 3.1.7 / Airbot

If you enable Airmode and Anti Grav on a switch, blackbox viewer incorrectly reports this as MAG and SUPER_EXPO

I assume the enum got out of sync when boris added BOXANTIGRAVITY at position 5 in the betaflight project, and this project's enum definition is out of date? 

It seems there are a few inconsistencies from my cursory inspection (does Super_Expo even exist as a mode change?)

Suppose it must be tricky managing multiple versions of the same enum.... perhaps you should only add values to enums unless doing a major refactor?

[BLACKBOX_LOG_Realacc210_20170606_095514.zip](https://github.com/betaflight/betaflight-configurator/files/1053729/BLACKBOX_LOG_Realacc210_20170606_095514.zip)
[image]

[image]

mikeller, 2017-06-06:
Duplicate of https://github.com/betaflight/blackbox-log-viewer/issues/60 . Also, please note that blackbox log viewer issues should be reported on https://github.com/betaflight/blackbox-log-viewer/issues .

closed 2017-06-06 by mikeller

rodgerramjet, 2017-06-06:
Sorry @mikeller  I looked for a dupe and everything - stupid me in wrong project :(