betaflight/betaflight-configurator #529: [Feature request] Disable support for newer 'development' Firmware in older BF Configurator releases.

richard-scott, 2017-06-18:
I have noticed a lot of users are using the latest v3.2 Firmware direct from GitHub... I know I am :-)

That's great and all, but it is creating a lot of issues for users where things just don't work quite right when using the v3.1.1 BF Configurator on a v3.2 Firmware. Things have changed in the newer BF v3.2 release so they just don't get configured anymore using the older Configurator.

It would be great if the BF Configurator refused to talk to any newer version of the Firmware. For example when using the v3.1.1 version of the BF Configurator it should only support 3.1.x Firmware releases. I would not suggest locking this down to the full version number (e.g. v3.1.1) as the BF Firmware is currently at v3.1.7 so that would cause issues.

When an unsupported Firmware version is detected the BF Configurator could then direct users to a Wiki page detailing the "Alternative way" to use the Chrome extension. This would a) save support questions about why things don't work the way you think it should, and b) enable some statistics gathering on users visiting the URL for the "Alternative way" documentation (if that is required by the project).

mikeller, 2017-06-19:
Hi @richard-scott. You are right, the current situation with respect to the configurator is a bit of a contingency. Normally, we fix this by periodically pushing updates of the configurator to the users, in order to make it work for users that are on the latest unreleased firmware (but also to get bugfixes and improvements to users with released versions of the firmware).
At the moment, we are not doing this because the latest version of the configurator in the repository breaks the battery configuration for users of released versions of the firmware, and there is no easy way to fix this.
For this reasons, users of unreleased versions of the firmware have to stick to the unreleased version of the configurator. This will be fixed when we push the next update for the configurator, around the time when Betaflight 3.2 is released.

richard-scott, 2017-06-26:
Yes but what about next time the 'stable' BF Configurator becomes incompatible with the ever advancing 'development' BF Firmware? I feel that version lock-in in hard coded into the BF Configurator would be a way to force people to use the 'development' BF Configurator with the their 'development' FC firmware?

Maybe you could link this up with the version bump thread #532 and get together with the BF Firmware guys to discuss a way to jointly and uniquely identify development versions?

I know that this would have saved me a couple of months time Googling for answers as to why my hardware wasn't working correctly... I want to try and help others not go down the road that almost put me off using BetaFlight.

mikeller, 2017-06-26:
The problem with this is that - as you say - the firmware moves very fast. Part of making this possible is that for a lot of the changes in the firmware (many of which are breaking some aspect of the configurator) we are not forcing the contributors to submit a PR for a configurator update at the same time. If we did, this would slow down firmware development, and introduce a lot of unneeded churn into the configurator.

In general, if you are using unreleased software, you should be prepared for everything not to work 'out of the box'. If you'd used a released version of Betaflight, this would not have been a problem. (Also, n.b., everything was always configurable just fine in the CLI, which is the case in general for Betaflight.)

closed 2017-07-14 by richard-scott