betaflight/betaflight-configurator #536: Motors keep spinning after leaving motor tab

Dutch-Drone-Dude, 2017-07-01:
I noticed that the motors keep spinning if you set the master slider at any given setpoint and you leave the motors tab. They even keep spinning if you disconnect from the board.

mikeller, 2017-07-01:
Yes. That is by design. You might want to go to the 'sensors' tab and look at some debug output while your motors are spinning, for example.

closed 2017-07-01 by mikeller

Dutch-Drone-Dude, 2017-07-01:
Okay I can imagine that, but even if you hit the disconnect button the motors keep spinning, isn't that a bit tricky?

mikeller, 2017-07-01:
Why? Your props are off. You might want to test something that requires disconnecting your craft from your PC.

Dutch-Drone-Dude, 2017-07-01:
Okay no problem, I thought it was an extra safety.

mikeller, 2017-07-01:
The safety is in removing the props. If a user does not do that despite explicitly confirming that they've done it, any assumption about them doing reasonable things is void anyway.