betaflight/betaflight-configurator #540: motors twitching when disarmed on multishot

killaGFPV, 2017-07-08:
bf config 3.1.2
bf 3.2.0
spf3
littlebee 35a 32bit esc's 
br2306 2700kv
matek pdb 4s

multishot motors twitch quite a bit and motors are noisy, 
d-shot 150/1200 issue. number 4 motor has a hard time spinning. airmode must be on to allow even rpm for take off. 
proshot1000, no result. no response from esc.

mikeller, 2017-07-09:
This does not sound like a configurator issue. Also, please note that GitHub is for discussion and questions around development of Betaflight. For configuration questions please go to RCGroups: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

closed 2017-07-09 by mikeller

richard-scott, 2017-07-13:
I had this... to get round it I went in BF Configurator Reciever TAB set Stick Min to be just above the lowest value on throttle channel with the stick as low as it can go. If this value is not less than the Min set in your ESC then change your TX to lower this value. I have mine at 996 as 1000 is the minimum throttle value on my ESC.

AndersHoglund, 2017-07-13:
Please, Tx/Rx throttle signal range is completely unrelated to the signal range seen by the ESC. 
If an analog ESC protocol is in use, the ESC signal range is ONLY the values set in `min_command` and  `max_throttle`. With an idle value controlled by `min_throttle` when armed.

killaGFPV, 2017-07-13:
thanks! i did that just yesterday, im still having issues with dshot600 and
littlebee 35a 32bit esc's one motor refuses to spin. is this an f3 board
issue? i built an f4v5pro with omnibus f4 3.2.0 and it works fine on dshot
1200. although i cannot see any of the gui for OSD in betaflight
configurator. i can however see the gui in cleanflight with betaflight
installed. but setting changes are not saved. the f4v4pro on dshot 1200 did
however flip out on me once when i punched and zero throttle stall and
punched again from 0rpm.