betaflight/betaflight-configurator #542: Regression: OSD tab blank for Betaflight 3.2/Omnibus F3 target

jrussellsmyth, 2017-07-08:
This problem is introduced by the merge of https://github.com/betaflight/betaflight-configurator/pull/534

OSD tab is blank for Betaflight 3.2 on Omnibus F3 target (possibly others)

Console log showing console error message and thrown exception.
[osd-tab-blank.log.txt](https://github.com/betaflight/betaflight-configurator/files/1133381/osd-tab-blank.log.txt)

mikeller, 2017-07-09:
Works all fine for me with an OMNIBUS F3, latest Betaflight, and latest configurator from GitHub. Do you get an error in the console?

jrussellsmyth, 2017-07-09:
Error is attached to the issue.

mikeller, 2017-07-09:
Ok. Looks like your firmware is older than your configurator. Can you update to latest firmware from GitHub and check again.

jrussellsmyth, 2017-07-09:
Updated to todays build, all is good. My apologies for the wild goose chase.

closed 2017-07-09 by jrussellsmyth