betaflight/betaflight-configurator #551: Font issue BF OSD (Betaflight font)

Arakon, 2017-07-16:
The "E" letter has a black pixel in the top left of it. Shows up both in the configurator and on-screen.
This is using the latest 3.1.2 configurator.

[image]
[image]

basdelfos, 2017-07-16:
Fix is added to PR: https://github.com/betaflight/betaflight-configurator/pull/553

basdelfos, 2017-07-16:
If you can't wait, download the font here and upload it to your FC ;)
https://drive.google.com/file/d/0B-6fyO8R3B_YSVFVbE1IT1ZTTms/view?usp=sharing

closed 2017-07-27 by jflyper