betaflight/betaflight-configurator #560: Request: Enable/Disable all button for OSD screen

Arakon, 2017-07-21:
Either that, or leave most of the uncommon options disabled by default.
Right now, when flashing a fresh firmware, almost everything is enabled and takes ages to disable one by one.

tigert, 2017-07-29:
I don't think it is possible to have a default setting that works for everybody, since each person has their own favorite setup.

But you really should use the "diff" command on CLI to save your configuration between firmware upgrades, that makes it easy to get your config back after a new version.

Arakon, 2017-07-29:
I do that, but when you do a clean setup, it's annoying.
Also, for some reason the "Horizon sidebar" setting doesn't seem to save in the diff command.. it's always enabled when I paste my commands, even though it's always disabled.

spuder, 2017-08-09:
A button to disable everything would be helpful. 

[image]

PR #691 (2017-10-16): Set Switch all checkbox defaults based on the majority of the switches
Fixes #560, #686
closed 2017-10-16 by mikeller