betaflight/betaflight-configurator #568: 3.1.2 can't manage motor tab

ChrisHammond, 2017-07-29:
I hate to open an issue for this, but at this point I have been unable to use 3.1.2 on a macbook, here's what I'm seeing.

Plug in Omnibus F3, switch to Motors tab, and the arm switch will be flipped on, motors show that they are on, but no battery hooked up and nothing spinning. Can't move the sliders.

Sometimes when I reboot, go to the motors tab, the switch will be office, but when I arm it, I have no control over the sliders.

Tested with two previously working FCs.

Plug the same FC into a PC that is still running 3.1.1 and everything works 100% as expected.

I'm hesitant to upgrade the PC to 3.1.2, but not sure if I can prevent that either with the Chrome extension.

ChrisHammond, 2017-07-29:
Well, I restarted Chrome on the PC and am getting the same result now. 

Start BFC
connect the USB Cable
Press the connect button
Navigate to Motors tab.
Arm toggle is on, 
Sliders don't work (screenshot 1)
[image]
Press disconnect button
Reconnect with the button
Navigate to the motors tab
Arm toggle is off, toggle it on
Sliders don't work
[image]

ChrisHammond, 2017-07-29:
Perhaps of note, I do have my motors reversed. I noticed this release has a new toggle switch for that. I'm unable to toggle that off and save (it always comes back as reversed after reboot)

jflyper, 2017-07-29:
If you have flashed your OMNIBUS with something that says Betaflight 3.2, then you should be using up to date configurator.

If this is not the case, we need to do some investigation.

ChrisHammond, 2017-07-29:
I haven't flashed off of 3.1.7 yet, using configurator 3.1.2

jflyper, 2017-07-29:
That is a problem.

Go back to BFC 3.1.1.

wind0r, 2017-07-29:
@jflyper  the big problem here is that 3.1.2 is in the appstore now.
we can not force every user to downgrade

Arakon, 2017-07-29:
Uh, why? 3.1.2 is not supposed to be released, since it goes with BF 3.2.0, which is also not released.
TONS of features don't work with 3.1.7 properly anymore.

ajimiii, 2017-07-29:
@Arakon my configurator is automatically updated to 3.1.2 too. How to downgrade to 3.1.1 since the chrome store only gives 3.1.2 configurator.

DanNixon, 2017-07-29:
You can download the 3.1.1 release here and add it to Chrome manually: https://github.com/betaflight/betaflight-configurator/releases/tag/v3.1.1

DanNixon, 2017-07-29:
3.1.2 is released prior to the RC being released to ensure people are using the correct configurator version when they come to install the RC firmware.

The fact it was released is not the issue, the fact it breaks compatibility with previous versions is.

commit ca39c713 (2017-07-30): Fixed motor tab for BF 3.1.x. Fixed #568
closed 2017-07-30 by borisbstyle