betaflight/betaflight-configurator #569: Beeper config not working

Arakon, 2017-07-29:
The buttons are there, but they don't do anything. Always reset to everything active in the configurator after save and reboot, while not actually modifying any of the beeper settings (checked in CLI). PR #563

basdelfos, 2017-07-29:
The changes in BF has to be merged yet and therefor it doesn't do anything at this point. Before official RC1 release this will be implemented and working.

closed 2017-07-29 by basdelfos

adfayed, 2020-09-15:
Hello, I am here running:
```
Configurator: 10.7.0
Firmware: BTFL 4.2.2
Target: IFRC/IFLIGHT_F722_TWING(STM32F7X2)
```
and I am still getting the same issue. Even after executing save in CLI mode it will **not** save beeper config. On reboot, everything resets to active.