betaflight/betaflight-configurator #570: Feature req: make "expert mode" toggle in configurator sticky

tigert, 2017-07-29:
It would probably be good to have the expert mode toggle remember its state after you close and reopen the configurator.

wind0r, 2017-07-29:
this option allready exists. just click the option symbol near expert mode. top right corner

Arakon, 2017-07-29:
It's reset next time you start the configurator, though.

wind0r, 2017-07-29:
Huh?
"Permanently enable Expert Mode" will not reset!

tigert, 2017-07-29:
Oh, duh, that was pretty hidden :-) Is there a reason the sticky setting it is split into a separate config option and not just save its state between configuration "sessions"?

wind0r, 2017-08-01:
Because there is the stick feature for experts and for normal users it should not be sticky. They need to know those settings are for "experts"

closed 2018-05-13 by stale[bot]