betaflight/betaflight-configurator #573: Switch to reverse motors is permanently on

pechovic, 2017-07-29:
The switch that is located right under the motor layout in the Configuration tab is permanently turned on. If you switch it off and reconnect your board, the switch will be again on. It doesn't reflect the actual value in betaflight because the motors spin the correct direction (not reversed)

pechovic, 2017-07-29:
Attaching a screenshot so that it is more obvious. Again the motors spin in the correct direction even though the configurator says they are reversed

[image]

Kasuhico, 2017-07-29:
I also noted the same issue... I was thinking that it was my FC problem, but I detected it in 2 different Omnibus V2 FCs. Good to know that it is a Configurator issue in this version released yesterday.

basdelfos, 2017-07-29:
That's on BF 3.1.7 right? It's fixed in PR #571

Kasuhico, 2017-07-29:
Correct, I'm using 3.1.7

pechovic, 2017-07-30:
@basdelfos yes BF 3.1.7

Thak you very much for the fix. I see the PR is not merged yet. So, once it is merged and released, I can test it and close this issue.

https://github.com/betaflight/betaflight-configurator/pull/571

basdelfos, 2017-07-30:
OK, that was een known compatibility issue with BF 3.1. We just pushed a new configurator (3.1.3) to the chrome store to with this issue.

closed 2017-07-30 by basdelfos

mikethecrash, 2017-07-30:
I'm having this problem and I can't seem to get the new version f the configurator, it's stuck on 3.1.2

Glad I found this thread though, I've being going nuts today!!

basdelfos, 2017-07-30:
Did you install the configurator manually? Then the auto update will not work. Try to remove the app from chrome and reinstall it from the chrome store.

mikethecrash, 2017-07-30:
yes that fixed it thank you :) 
How to install the configurator non-manually so that the auto-update works?
thanks again

basdelfos, 2017-07-31:
Remove the manually installed app from chrome, type in URL: chrome://extensions/
Remove it by clicking on the recycle bin icon.

Install official released app from the chrome store, type in URL: https://chrome.google.com/webstore
Search for Betaflight and install the configurator. Now it will auto-update with every official release.

mikethecrash, 2017-07-31:
OK will try, thanks.