betaflight/betaflight-configurator #579: CPU Load shows %100 after changing gyro settings

devbloggs, 2017-07-30:
Whenever changing the gyro freq once saved and the board reboots, the cpu load is stuck at %100 until the configurator is closed and then re-opened
[image]

Arakon, 2017-07-30:
Your configurator is outdated, 3.1.2 has issues with pre-3.2.0 firmwares. Update and check again.

devbloggs, 2017-07-30:
ive updated to 3.1.3, still the same issue

basdelfos, 2017-07-30:
I can't reproduce this issue on 3.1.7 and configurator 3.1.3 with on an OMNIBUSF4.
Someone else with MKF4 target can reproduce this?

devbloggs, 2017-07-30:
I have just tried 3.1.3 configurator on a different pc, with another board of the same mkf4 target with the same results :$

Bill-Glover, 2017-08-04:
I have this issue with the new v3.2.0 configurator and BF 3.1.7 (BF F3 target). It's consistent and reproducible - any change to PID or gyro causes CPU to show as a constant 100% after save & reboot. Hitting disconnect then reconnect fixes it.

swilkens, 2017-08-04:
I checked and verified on my own BF F3 board running 3.1.7 and the 3.2.0 chrome store configurator.

It seems the SDCARD config switch is involved, here is my method to trigger:
1. Flash a clean 3.1.7 to the BF F3 board
2. Connect to the board using 3.2.0 configurator
3. Change the Gyro update speed, click save and reboot
4. Observe 100% CPU Load in the configurator
5. In the CONFIGURATIONS tab, disable the SDCARD feature, click save and reboot
6. Observe normal CPU load in the configurator
7. In the CONFIGURATIONS tab, enable the SDCARD feature, click save and reboot
8. Observe 100% CPU Load in the configurator once again

All the while, the actual load on the F3 chip seems normal.

[image]

swilkens, 2017-08-04:
Update: 
In fact, JUST switching ON the SDCARD and hitting save & reboot will trigger this. Gyro / PID changes are not needed to trigger it.

closed 2018-05-13 by stale[bot]