betaflight/betaflight-configurator #580: 3.1.2 "reversed motor direction" is always checked when it's not

timothyswt, 2017-07-30:
I have a F4 V5 PRO that when its connected to betaflight, on the configuration tab it shows that the mototor direction is reversed:
[image]

except that it is not:
[image]

and every time I try to set it, it still shows reversed.

pechovic, 2017-07-30:
Duplicate. https://github.com/betaflight/betaflight-configurator/issues/573

Should be closed.

timothyswt, 2017-07-30:
Sorry I didn't see that one. I did a search but I just have done a poor job searching. Thanks!

closed 2017-07-30 by timothyswt