betaflight/betaflight-configurator #584: Profiles don't always change in configurator 3.1.2

aowi7280, 2017-07-31:
When selecting Profiles and rate profiles they don't get selected. Ex. Starting at Profile 1 Rateprofile 1, then I select Profile 2, rate profile 2. When I try to go back, it stays in Rateprofile 2.

basdelfos, 2017-07-31:
Please upgrade to the newest configurator 3.1.3 from the chrome store and check if the issue still exist.

s700ky, 2017-08-01:
My Chrome config is 3.2 

When selecting profile 2 or 3 the profile drop down goes back to 1. 

I can’t set profiles unless I go via the CLI.

basdelfos, 2017-08-02:
Which BF firmware are you using?

s700ky, 2017-08-02:
3.1.7

chime13, 2017-08-02:
Same here - Since 3.1.2 the Profiles/Rateprofile selections have been broken. I'm running 5 FCs on BF 3.1.5, 3.1.6, 3.1.7. All exhibit the same issue - Rateprofile stuck at 1 and Profile 2 is unselectable

s700ky, 2017-08-02:
I upgraded to Betaflight 3.2 RC2 and profiles work fine.

basdelfos, 2017-08-03:
I can reproduce this. I will try to fix this before I go on holiday next
week.
If you look at the log in the configurator you will see that it activated
the profile/rateprofile you have selected, but the dropdown box switches
back to the initial selected profile/rateprofile. Very strange that this
doesn't happen on 3.2-RC2.

Also, it takes 1-2 seconds to update the GUI after changing the
profile/rateprofile. When changing the profile/rateprofile again during the
time the GUI is updating this makes it even more worse.

chime13, 2017-08-05:
Greetings @basdelfos,
I downloaded 3.2.RC2 to see if the issue was resolved. Unfortunately, the problem still exists.
GUI defaults to the last RateProfile used on the FC. 
For example: 
So, let's say the last profile I used was "2" with Rateprofile 3. When I load the PIDS tab in BFC, the boxes will correctly read Profile 2 and Rateprofile 3. However, since 3.1.2 - In my example, If I change to Rateprofile 2, the log will update to Rateprofile 2, but the GUI will still read Rateprofile 3 **_AND_** Rateprofile 2 is no longer selectable. 1 and 3 still update in the log but 2 cannot be addressed.
In other words - Once you change to a different Rateprofile, the Rateprofile that was initially loaded is no longer available and does not reflect in log.
Hope this helps!
Enjoy your holiday!!!

closed 2017-08-06 by borisbstyle