betaflight/betaflight-configurator #588: Save flash to file not working

TheIsotopes, 2017-08-01:
using latest bf 3.2.0 rc2 on revolt f4 and latest master from bf configurator ... but now i can not more save bblogs ... if i'm klick `save flash to file` the files are 0 bytes

@martinbudden pls can you check this ... before #549 there was not this issue

DieHertz, 2017-08-01:
Try saving flash to file again, then open up Chrome Console for BFC and copy it's output to this  discussion.

TheIsotopes, 2017-08-01:
@DieHertz here the output from console

```
main.js:72 Libraries: jQuery - 2.1.4, d3 - 3.5.6, three.js - 72
port_handler.js:71 Selecting last used port: COM7
serial_backend.js:49 Connecting to: COM7
serial.js:134 SERIAL: Connection opened with ID: 1, Baud: 115200
three.min.js:536 THREE.WebGLRenderer 72
main.html#:1 [.Offscreen-For-WebGL-00000265F9B2C0F0]GL ERROR :GL_INVALID_ENUM : GetIntegerv: <- error from previous GL command
main.html#:1 [.Offscreen-For-WebGL-00000265F9B2C0F0]GL ERROR :GL_INVALID_ENUM : GetIntegerv: <- error from previous GL command
main.html#:1 [.Offscreen-For-WebGL-00000265F9689EA0]GL ERROR :GL_INVALID_ENUM : GetIntegerv: <- error from previous GL command
main.html#:1 [.Offscreen-For-WebGL-00000265F9689EA0]GL ERROR :GL_INVALID_ENUM : GetIntegerv: <- error from previous GL command
main.html#:1 [.Offscreen-For-WebGL-00000265F9689EA0]GL ERROR :GL_INVALID_ENUM : glTexImage2D: <- error from previous GL command
three.min.js:602 THREE.WebGLProgram: gl.getProgramInfoLog() 
C:\fakepath(106,3-100): warning X3557: loop only executes for 1 iteration(s), forcing loop to unroll

 
(anonymous) @ three.min.js:602
onboard_logging.js:452 Dataflash dump file path: D:\Modellbau\RaceCopter\BTFL_BLACKBOX_LOG_HX200_20170801_113707.BBL
onboard_logging.js:343 Received 0 bytes in 0.08s (0.00kB / s) with block size 4096.
onboard_logging.js:452 Dataflash dump file path: D:\Modellbau\RaceCopter\BTFL_BLACKBOX_LOG_HX200_20170801_113835.BBL
onboard_logging.js:343 Received 0 bytes in 0.07s (0.00kB / s) with block size 4096.

```

DieHertz, 2017-08-01:
Nothing wrong in here. I can't really look into this before I get home.

DieHertz, 2017-08-01:
Indeed flash download works on F3 but not on F4 with compression, trying to figure it out.

TheIsotopes, 2017-08-02:
used latest jenkins build 1540 and now it works ... thx 👍 ... closed now

closed 2017-08-02 by TheIsotopes